Vlsir/Vlsir #41: Some means of automated testing `vlsirtools.spice`

dan-fritchman, 2022-12-14:
We brick these simulator-drives more often than seemingly anything else, and for good reason: we don't really test them. 

Outlook for doing so: 

* Anything with a closed-source license basically won't happen, unless it's stood up separately on privately owned hardware
* Xyce installation is: 
  * Very easy on MacOS, CentOS/ RedHat Linux
  * Very, very painful on Ubuntu/ Debian Linux. Like a full-day project. 
  * Relevant: our CI provider (GitHub Actions) supports Ubuntu, but not CentOS/ RedHat. 
  * Probably works on Windows? Idunno anything about Windows.
* NgSpice is quite installable on all of the above, but is also in many respects "the worst" of these, both in and of itself, and in our support for it. 

Writing this makes me realize: we probably need something like the OS-testing matrix, or at least to swap out Ubuntu for testing Xyce.

dan-fritchman, 2022-12-14:
Related: unit testing for each simulator essentially runs "if the simulator's available". Tests generally look like so: 

```python
@pytest.mark.skipif(
    not vlsirtools.xyce.available(),
    reason="No Xyce installation on path",
)
def test_xyce1():
    # ... do stuff ....
```

We would seem to want a test-option that says "such-and-such simulator **better be** available, or fail". This could be in the form of custom pytest command-line options, such as those deployed [here](https://github.com/Vlsir/Usb2Phy/blob/main/Usb2PhyAna/conftest.py).

ThomasPluck, 2023-06-18:
Question: are we going to close this one when we have testing across Windows, MacOS and a Linux?

dan-fritchman, 2023-06-19:
I think we should close this now, and should have once you added the simulation-focused test container!

closed 2023-06-19 by dan-fritchman