Vlsir/Vlsir #49: The "Official Language Bindings" Story

dan-fritchman, 2023-04-10:
The issue: sort out which languages/ bindings should be kept in-source as part of VLSIR.  
(@growly thinks "none", we know.)

There are two language that are kinda special, because we have co-authored downstream projects which depend on them: 
* VlsirTools depends on the Python bindings
* Layout21 depends on the Rust bindings

All the other languages are more debatable.  
The published C++ bindings, for example, are used by no one that we know of. Arya's BFG uses the schema, in C++, but compiles them in its own build process, which needs to be customized for a variety of reasons. The other languages are probably not in direct formal use anywhere. 

Proposal: we make the others "recommended recipes". We are not really using or testing them; we should not represent that we do. Building these "recipes" in CI seems a fine step, as does publishing the build script "recommendations". But we should probably drop the generated code itself.

growly, 2023-04-10:
agree 👍

dan-fritchman, 2023-04-27:
Done via #48

closed 2023-04-27 by dan-fritchman