Vlsir/Vlsir #50: Integrate Netlist into VlsirTools

ThomasPluck, 2023-04-21:
Citing https://github.com/dan-fritchman/Hdl21/issues/108, VLSIR should be expanded to integrate [Netlist](https://github.com/dan-fritchman/Netlist/tree/main/netlist) into VlsirTools to make it possible to read and simulate external netlists. I propose the following plan:

- [ ] Integrate Netlist as is into VlsirTools 
- [ ] Write conversion scripts to partially convert Netlist data model in VLSIR
- [ ] Expand VLSIR protocol buffers to include the Netlist data model
- [ ] Finish integrating Netlist into VlsirTools and parse all netlists to expanded VLSIR

dan-fritchman, 2023-04-21:
Sounds great. Recommend a few tweaks: 

* Put `Netlist` as another python-package, alongside `VlsirTools` (and `SpiceCmp`) for that matter
* I don't think we want to expand the VLSIR data model to include all the fun transistor-modeling stuff in `Netlist`, at least not yet. 
* Conversion into VLSIR will take a bit more than just a script. The `Netlist` parsing particularly needs to be expanded to do "resolution". I.e. in compiler terms, it currently makes the AST, and needs to make the CST for VLSIR to work. Example: 

```
* First subcircuit named "who_am_i"
.subckt who_am_i 
+ alpha beta gamma
.ends * who_am_i

.subckt outer a b c 
  * Another, inner-scoped subcircuit with the same name "who_am_i"
  .subckt who_am_i x y z 
  .ends * who_am_i

  * The trick: which subcircuit does this resolve to? 
  * The answer can often be "none", if the expectation is it's provided externally
  x1 a b c who_am_i

.ends * outer
```

To date each `Instance` just stores a string-valued `Module` (or similar). They'll need to be resolved, sometimes nestedly as in the example above. And the ones that can't be resolved can create `ExternalModule`s on the fly.

ThomasPluck, 2023-04-22:
Okay, so here is the tweaked task list:

- [ ] Add `Netlist` as a standalone Python package, alongside `VlsirTools` and `SpiceCmp`.
- [ ] Expand the `Netlist` parsing process to perform "resolution" to construct a Spice CST.
- [ ] Develop a method to convert from a Spice CST into a partial `VLSIR` representation.

ThomasPluck, 2023-04-23:
So, I've taken a look at Netlist today and I believe as a first PR that the CST generation for the circuit description portions of a netlist are adequate to convert to VLSIR barring the pathological case above. But that is as far as I'll go, because I think there needs to be a more involved strategy, if any, to get simulation keywords correctly parsed out and converted to VLSIR. I'll then double back to dan-fritchman/Hdl21#108 and implement a few ways to go `from_proto` into the specific PDK ExternalModule libraries or native HDL21 primitives.

dan-fritchman, 2023-06-28:
So on 2nd (3rd? 4th? whatever) thought here - 

* There's already lots of repetition between `Netlist` and `VlsirTools` - particularly for *writing* netlist-text. 
* It breaks plenty of the time, requires maintenance etc. 
* That would be good to centralize. 

So perhaps: 

* `Netlist` becomes the (sole) writer for the two
* `Netlist` retains its very-SPICE-specific data model 
* `VlsitTools`, instead of writing netlist-text, would primarily convert its stuff into the `Netlist` data model, and invoke its writer(s).

The one thing currently supported by VlsirTools that *might* get left behind is Verilog. That language has a lot more "stuff" in the data model, some of which maps more cleanly onto VLSIR schema than the spice-centric `Netlist` data model. VlsirTools would probably still need to implement that one itself.