Vlsir/Vlsir #56: Fix asyncio embedding issue

uduse, 2023-05-01:
As mentioned in: https://github.com/dan-fritchman/Hdl21/issues/113?notification_referrer_id=NT_kwDOAEf5zbI2MjEwOTk5MDc5OjQ3MTcwMDU#issuecomment-1528319399

https://github.com/Vlsir/Vlsir/blob/e7f5f91a0a4e14469cd33c6a5c3b9b1b321a6058/VlsirTools/vlsirtools/spice/spice.py#L85 should be updated to detect existing loop.

Also can someone add me as a collaborator for easier PR?

uduse, 2023-05-16:
> You're in

Hm... it seems I still can't push directly in this repo. I am in Hdl21 though.

dan-fritchman, 2023-05-16:
Hm indeed.  
I think you're in (again).

dan-fritchman, 2023-05-19:
OK this seems bad: 
https://github.com/erdewit/nest_asyncio/issues/57

Especially in the context of calling simulations, which fail and generate exceptions all the time.

dan-fritchman, 2023-05-19:
So it seems given this limitation, one can have either of 

* (a) Users must know whether they are in an `async` or synchronous context, and call `sim` or `sim_async` accordingly, or
* (b) They can get exceptions 

Given the two options, I think I'd pretty easily choose (b). 

It does bring to mind, what is all this `async` stuff _for_. The only actual parallelism we got out of it, is while waiting for external subprocesses to complete. All the python-code runs in a single thread, blocking-ly. There's likely a subprocess-equivalent of `asyncio.gather`, or a custom version to be written here, that gets rid of this async-stuff. Now `async` is _nice_ because we don't need to write the state-machine that polls each subprocess before, for example, reading result-files. But we could.

dan-fritchman, 2023-05-19:
BTW as far as the Jupyter use-case goes - it appears that nowadays you can `await` directly in the cells: 

[image]

Granted this would still fail if, for example, you imported a python-module which invokes the synchronous `sim` function and `asyncio.run`.

uduse, 2023-05-20:
I think asyncio is mainly designed for use cases where the users expects async behavior. e.g., dealing with networks. A library using asyncio should not have an `asyncio.run`, since it should be the responsibility of the user to call `asyncio.run` on the top of the user's program.

uduse, 2023-05-20:
asyncio should "contagious" sort to say. If a lib uses async functions, other libs that build on top of it should use asyncio too. That's one big con of cooperative multitasking, that everyone has to cooperate :p

uduse, 2023-05-20:
Given my experience of using asyncio in various projects, seeing people using it, and reading relevant docs online, people are generally clueless so it's hard to build a library using asyncio and hope the user "cooperate". 

> Users must know whether they are in an async or synchronous context, and call sim or sim_async accordingly

Always exposing `async def` interface and ask for explicit usage might be a good idea since it's like say "use only if you know what you are doing".

dan-fritchman, 2023-05-22:
> I think asyncio is mainly designed for use cases where the users expects async behavior. e.g., dealing with networks. A library using asyncio should not have an asyncio.run, since it should be the responsibility of the user to call asyncio.run on the top of the user's program.

> asyncio should "contagious" sort to say. If a lib uses async functions, other libs that build on top of it should use asyncio too. That's one big con of cooperative multitasking, that everyone has to cooperate :p

> Given my experience of using asyncio in various projects, seeing people using it, and reading relevant docs online, people are generally clueless so it's hard to build a library using asyncio and hope the user "cooperate".

Seems right on all counts.  
Async, at least as it stands today, has gotta be an "application level" decision. And this is a library intended for many diverse applications, many of which will be quite clueless about async.  
We should probably design it out.

uduse, 2023-05-22:
Agreed that it should be removed. Fortunately we don't use it much so changes are quite local.

closed 2023-06-22 by dan-fritchman