Vlsir/Vlsir #61: Non-async simulation sub-process

dan-fritchman, 2023-05-22:
To date we have used two methods for launching batches of `vlsirtools.spice` simulations: 

* (a) Originally, run subprocesses synchronously, blockingly, and in series
* (b) Use Python's `async` infrastructure, particularly `async.subprocess`

Option (a) had the obvious disadvantage of being, well, slow. 
And #56 details the problems using `async` in our context. 
I think those problems are bad enough that we should either go back to (a), or design something to replace (b).

## Potential Replacement 

At its core this will ultimately be a pile of `SubProcess`es. Ideally we can also interleave things that happen both after they complete (i.e. results parsing) and before they start (i.e. elaboration) with waiting for them to complete. Note as in the `async` use-case, the only *parallelism* here is between sub-processes; all Python code runs in a single thread, blockingly and synchronously. 

So, something (at least conceptually) like: 

```python
class State(Enum):
  WAITING = 0 # for a subprocess to complete
  READY = 1 # for post-processing
  DONE = 2 # with post-processing

@dataclass 
class Sim:
  subproc: subprocess.SubProcess  # The simulator subprocess
  after: Callable # Some code that runs after the subprocess
  state: State # 

def launch_em(sims: List[Sim]):
  for sim in sims: 
    # Get each to the `WAITING` state, i.e. to the point its subprocess is running 
    sim.do_that_stuff_in_that_comment()

def manage_em(sims: List[Sim]) -> bool:
  # The boolean return value is "are they all done"

  # Get the sims in the READY state, run their `after`, and move em to done
  ready = [sim for sim in sims if sim.state == State.READY]
  for sim in ready: 
    sim.after()
    sim.state = DONE

  # For each in the WAITING state, do some subprocess stuff to see if it can be upgraded to READY 
  waiting = [sim for sim in sims if sim.state == State.WAITING]
  for sim in waiting:
    if some_subprocess_stuff(sim):
      sim.state = READY 

  return any([s.state != State.DONE for s in sims])
```

And then that `manage_em` loop would sit in our home-grown "event loop", which would be something like: 

```python
def event_loop():
  while not manage_em():
    time.sleep(some_amount)
```

Those can be added to the existing `Sim` classes as attributes, or made as new things which wrap them.

dan-fritchman, 2023-06-14:
@ThomasPluck you interested in this one? 
This, #56, and https://github.com/dan-fritchman/Hdl21/issues/122 are all kinda one meta-issue. 

Thinking about it slightly more - the concept outlined above would work great for our Nsgpice and Spectre drivers.  
For Xyce, it's a little different, as we launch separate subprocesses *per analysis*. That in turn is because we've had so much trouble parsing results from multi-analysis Xyce sims. There are a few options I can imagine: 

* All of the analysis-sub-processes go into the "pool"; we continue to parallelize across them. Fastest, and breaks that model above the most
* Xyce just works sequentially, at least for now
* Maybe just throw the *last* analysis in the process pool, after running the others synchronously?

ThomasPluck, 2023-06-14:
Sure, I can take a look tomorrow evening!

closed 2023-06-22 by dan-fritchman