Vlsir/Vlsir #63: Netlisting `SpiceType`s

dan-fritchman, 2023-06-07:
The VLSIR half of https://github.com/dan-fritchman/Hdl21/issues/135

dan-fritchman, 2023-06-07:
So we've kinda defined two ways to produce a few kinds of spice primitives. 

The way `vlsir.primitives` works is: 

* There is a generic `ExternalModule` named e.g. `Mos`
* It has a (instance) parameter named `modelname`

```python
        ExternalModule(
            name=_qname("mos"),
            desc=dedent(
                """
                # Mosfet Transistor

                Ports: (d, g, s, b), in identical order to SPICE convention 
                Params: string modelname

                `vlsir.primitives.mos` Largely corresponds to the "M-prefix" element of Spice-class simulators. 
                Each instance maps to a *spice model* instance ("`m1`"), *not* to sub-circuit instance ("`x1`"). 
                In many cases, particularly for technology-provided devices, using a foundry-provided sub-circuit will be appropriate instead. 

                The *sole* required parameter to each `vlsir.primitives.mos` is its string-valued `modelname`. 
                Additional parameters such as physical dimensions are typically model-specific and vary widely between models. 
                Conversion from `vlsir.primitives.mos` to spice-netlist formats is to pass all such additional parameters unmodified. 

                Example instantiation:

                ```python
                Instance(
                    name="mos1",
                    module=Reference(domain="vlsir.primitives", name="mos"),
                    connections=dict(d=d, g=g, s=s, b=b),
                    parameters=dict(
                        modelname="my_favorite_nmos", 
                        w=1e-6, 
                        geomod=2, # An example highly model-specific parameter
                    ),
                )
                ```

                Corresponds to netlist-level content along the lines of: 

                ```spice
                .model my_favorite_nmos     * Model statement provided externally 
                + nmos level=53             * Model parameters

                * Instance compiled from `vlsir.primitives.mos`:
                mmos1                       * Note the `m` prefix
                + d g s b                   * Connections 
                + my_favorite_nmos          * Model name
                + w=1e-6 geomod=2           * Instance parameters, unmodified
                ```
                """
            ),
            ports=_ports(["d", "g", "s", "b"]),
            signals=_signals(["d", "g", "s", "b"]),
            parameters=[
                Param(name="modelname", desc="Model Name (string)"),
            ],
        ),
```

This `vlsir.primitives.Mos` is thus *kinda* more like an `ExternalModule`-*generator* than an ExternalModule itself. 

The new way is:

* You define any `ExternalModule` you like
* You give it an enumerated `SpiceType` attribute
* If that `SpiceType` is a primitive one - i.e. anything but `SUBCKT` - we (intend to) netlist it as that spice-primitive-type. 
  * This includes sticking the right prefix on any instance names, arranging parameters accordingly, etc. 

e.g. (using Hdl21's `ExternalModule`, which is more or less the same as `vlsir.ExternalModule`):

```python
    NmosModel = h.ExternalModule(
        name="nmos_model",
        spicetype=SpiceType.MOS, 
        ... 
    )
```

Now, it seems pretty clear that if we want *that*, we want it to netlist as a thing named `nmos_model`. 

Seems we could either: 

* (a) Special-case the "model based" `vlsir.primitives` external-module "generators" - `mos`, `bipolar`, `diode`, and `tline`
* (b) ... Deprecate those? Just use the new method instead?

dan-fritchman, 2023-06-09:
So, not that `sky130_fd_pr__res_generic_po` is the most important module in the world or anything - but I did use it as an example here. With the current head-of-main VLSIR and Hdl21, this bit of code here: 

```python
import sys
import sky130
import hdl21 as h
from vlsirtools.netlist import NetlistFormat

@h.module
class HasRes:
    s = h.Signal()
    r = sky130.modules.GEN_PO()(p=s, n=s)

h.netlist(HasRes, sys.stdout, fmt=NetlistFormat.SPICE)
h.netlist(HasRes, sys.stdout, fmt=NetlistFormat.XYCE)
h.netlist(HasRes, sys.stdout, fmt=NetlistFormat.NGSPICE)
h.netlist(HasRes, sys.stdout, fmt=NetlistFormat.SPECTRE)
```

Produces this bit of (slightly abbreviated) netlist: 

```
.SUBCKT HasRes
rr
+ s s 
+ w='1000m' l='1000m' m='1' 
.ENDS
```
```
.SUBCKT HasRes
rr
+ s s 
+ PARAMS: w={1000m} l={1000m} m={1} 
.ENDS
```
```
.SUBCKT HasRes
rr
+ s s 
+ w='1000m' l='1000m' m='1' 
.ENDS
```
```
subckt HasRes 
  r
  + // Ports: 
  + ( s s )
  +  resistor 
  +  w=1000m l=1000m m=1   
ends 
```

Which, I don't see how that's what we want, for any conceivable EDA tool. Notably, at no point do they mention wanting a `sky130_fd_pr__res_generic_po`. 

Now what *is* that `sky130_fd_pr__res_generic_po` thing? Appears it depends who (where) you ask: 

* [Here](https://github.com/google/skywater-pdk-libs-sky130_fd_pr/blob/f62031a1be9aefe902d6d54cddd6f59b57627436/models/r%2Bc.mrp1monte.spice#L28) it's defined as a subcircuit
* In my own local installation (via litex-hub conda), in `sky130_fd_pr__model__r+c.model.spice`, I see it defined as a `.model`. I have not been able to find this on GitHub, or otherwise track it back to source. 

```
.model sky130_fd_pr__res_generic_po       r tc1r=tc1rsgpu   tc2r=tc2rsgpu   rsh=rp1   dw="-tol_poly/2-poly_dw/2" tnom=30
```

Which does any given tool (much less version of this highly diffuse PDK) pick up? Great question! 

---

## Zooming Out

Rooting through this and the related models has (I think) offered some clarity on how it should be handled.  
Until the recent addition of the `SpiceType` field on user-settable `ExternalModule`s, we (on some level) could netlist instances of two kinds of things: 

* Spice primitives, which all came from `vlsir.primitives`
* Subcircuits, which come either from `vlsir.Module`, or refer to externally-defined subcircuits via `vlsir.ExternalModule`

A few of those `vlsir.primitives` always had something fishy hiding inside though - a `modelname` *instance* parameter, which really should have been baked into something more like the `Module` level. 

In reality I think these should be three kinds of things: 

1. Spice *builtins*, e.g. resistor, capacitor, vdc 
2. Spice *model references*, i.e. instances of things defined by something like the `.model` statement
3. Subcircuits/ Modules

And: 

* (1) continues to be defined by `vlsir.primitives`. 
* Any user-defined `ExternalModule` with a  non-`SUBCKT` `SpiceType` falls in bucket (2). (What else could a user-defined spice primitive *mean* after all?) These won't have any constraints on parameters, passing them along directly to instances. And I guess they won't have constraints on ports either(?). And they can use their `ExternalModule.name` as the specifier for the model-name. 
* Bucket (3) stays the same. 

This would kick the "model references" out of `vlsir.primitives`. They are the same things referred to as "model based" in the netlisting code to date: 

```python
model_based = {
        SpiceType.MOS,
        SpiceType.BIPOLAR,
        SpiceType.DIODE,
        SpiceType.TLINE,
    }
```

I think these should be deprecated. We can include "generator functions" which produce `ExternalModule` "like them" (I guess from just a string `name`) in `VlsirTools`, or with #57 in a new primitives-package. But they're more like "module generators" than they are "modules" themselves.

ThomasPluck, 2023-06-09:
> Is there some use-case (i.e. downstream EDA program) where you want these elements to be SPICE-primitive resistors, but not have a specified resistance? What is it?

As you've noticed, this is just how generic resistors are defined in the PDK, the "generic" sheet resistors are a `.model` who's parameters are then pulled into the resistor definition with its name, for instance:

In, `libs.tech/ngspice/sky130_fd_pr__model__r+c.model.spice`
```
.model sky130_fd_pr__res_generic_m1 r tc1r = {tc1rm1} tc2r = {tc2rm1} rsh = {rm1} dw = {"-tol_m1/2-m1_dw/2"} tnom = 30.0
```
Defines the M1 resistor model and its parameters are used to define the resistor that we want to see below in microns.
```
R1 a b sky130_fd_pr__res_generic_m1 W=1 L=1 m=1
```
So yeah, the model name *should* be included when defining our resistor.

## Zooming Out

So the story seems to be that `vlsir.primitives` are SPICE components with a modelname "hack" that keeps them behaving like SPICE, but we now have a more explicit way to tie things to models and so we're now saying that `vlsir.primitives` are to be "model-free" generic components.

Only comment I'd make here is if `.model`s are important enough warrant their own keyword in SPICE, then VLSIR and friends should reflect that so user's are guided by the data structures to a model being a significant thing that users can manipulate and use.

## Zooming Back In

Thinking now of implementation this follows the `to_proto` story in HDL21, in my mind. All PDK components are `ExternalModule`s and so pass along their `SpiceType` and `name` parameters to the `VlsirTools` netlister. From there, we'd just need to add an extra-line which inserts the `name` as a line on the SPICE file to simulate - simple!

dan-fritchman, 2023-06-10:
Yep. 
I think #63 handles this the right way now.

closed 2023-06-12 by dan-fritchman