Vlsir/Vlsir #85: Random Noise Sources

ThomasPluck, 2023-09-12:
At least in ngspice, you can include noise sources that can be used in ordinary transient analysis - is this widespread enough phenomena across simulators to warrant its own VLSIR binding?

dan-fritchman, 2023-09-12:
Can you really? 

Usually in SPICE, "noise" - i.e. the stuff the `.noise` analysis is analyzing - is coming from the *devices*, not the *sources*. The "noise voltage/ current sources" are essentially built in to, like, BSIM and other device models . 

Tran-noise is different in how it analyzes the circuit (i.e. large signal-ly), but no different in this respect. The "noise" remains that generated *unintentionally* (to you) by the resistors, MOSFETs, and the like.  

What is, like, the parameter-space for the NgSpice thing in question? Like, gaussian norm and mean? Or is it like a digital noise source, like `vpulse` but with random values?  
I don't know of any of the above in the big commercial tools; if there is one I'd be pretty confident it's not oft-used.

ThomasPluck, 2023-09-13:
Sure, there are two `TRNOISE` which mimic common physical noise like white, 1/f, random telegraph and `TRRANDOM` which brings the Gaussian, Poisson, Uniform, Exponential distributions to currents and voltages.

AFAIK, you can also give ngspice formulas to transform the sampling, so you could the full range of statistical distributions, if you really felt like it.

I'm coming from stochastic dynamics in math-land, so for me, noise is signal - it can be used to do stuff. I was hoping that these festures might be more pertinent to electronic people than it is. 

No problem in this case, I'll have these as external modules which I tack onto my workflow.

dan-fritchman, 2023-09-13:
Ok cool.  
I suppose more than anything I’m curious - how do you intend to use these things? Is there a test bench or simulation setup you could share?

ThomasPluck, 2023-09-13:
Sure, here's one I made earlier:

https://colab.research.google.com/drive/1b57Et1a-exLP1RB1dDJ0gXTg2tEuYmkB

ThomasPluck, 2023-09-13:
Actually, while we're on the subject, another thing that's going to need implementing in #84 is the Monte Carlo simulations - so the way this is done in ngspice is using `sunif`, `sgauss`, `poisson` and `exponential` functions which explicitly construct a Monte Carlo vector to run a simulation on.

So `VLSIR` in its schema has slots for `LinearSweep`, `LogSweep` and `PointSweep` - could there be something similar made for these "Monte Carlo functions" or is this something that ngspice does that is strange and should be left as "Custom" analysis?

ThomasPluck, 2023-09-19:
After talking with Dan yesterday, it turns out that ordinary simulators actually expose a friendly `.monte` command that can be used with ease by designers to run simulations on expected process variations, rather than ngspice's "bring your own Monte Carlo" approach.

Since, this aspect of ngspice doesn't translate well to other simulators, we agreed that it'd be best to leave Monte Carlo to custom ngspice analysis.

closed 2023-09-19 by ThomasPluck