Vlsir/Vlsir #99: new release supporting python3.12

joamatab, 2024-08-30:
Hi Dan,

can you make a new release for python3.12?

joamatab, 2024-12-09:
Hi Dan, 

Any updates on the release? :)

dan-fritchman, 2024-12-10:
No news here. 
I thought you opened a PR for this? But I don't see it.  

There are certainly a few Hdl21 PRs:
- https://github.com/dan-fritchman/Hdl21/pull/231
- https://github.com/dan-fritchman/Hdl21/pull/232

Which I believer were mired in some combination of Pydantic/ Python version compatibility.  
(There is commentary to that effect written somewhere I also can't immediately find.) 

Do you guys need Hdl21 or just VLSIR?

joamatab, 2024-12-10:
Just vlsir for now would be great to have :)

dan-fritchman, 2024-12-10:
Yeah that should be easier.  
Do you or someone from your team wanna put in a PR?

joamatab, 2024-12-10:
I think we just need to release to PyPi

we already merged this
https://github.com/Vlsir/Vlsir/pull/98

dan-fritchman, 2024-12-10:
Ah yes there it is!  
Yeah I got it.

joamatab, 2024-12-11:
How do you release to pypi? let me know if i can help with that

dan-fritchman, 2024-12-12:
It's now live: https://pypi.org/project/vlsir/7.0.0/

Along with Hdl21 v7.0, and all the associated PDKs & similar packages.

dan-fritchman, 2024-12-12:
Noting the CI tests with Python 3.12 continue to fail. I filed #101 for tracking getting them cleaned up. (Another welcome PR!) I am confident that is a problem with the test setup and not the package itself.