Vlsir/Vlsir #102: PyPi Org, Releases

dan-fritchman, 2024-12-12:
The "VLSIR family" has a half dozen or so interdependent PyPi packages: 

- vlsir - the schema bindings
- vlsirtools - most of the useful code
- hdl21
- each of the open PDKs: ASAP7, Sky, GF
- a few others which are internal or deprecated

We would like to enable some more contributors to help publish them. Especially as the maintainers focuses have shifted elsewhere. I think we want to: 

- Create a pypi.org organization
- Move all this stuff into it
- Grant one or more new folks publishing access

I have a request in for a `vlsir` org on pypi. They apparently review & approve each one.  
@joamatab - how are you guys doing this for GdsFactory?  
And do you have any other team members you'd like to have weigh in?

joamatab, 2024-12-12:
Hi Dan,

For the GDSFactory organization, we use a PyPI organization secret to handle publishing to PyPI seamlessly.

I’d be happy to assist with any improvements or updates you’re planning for VLSIR.

Here’s the GitHub Action workflow we currently use for our releases:
[Release Workflow](https://github.com/gdsfactory/gdsfactory/blob/main/.github/workflows/release.yml)

I've cc'ed Niko who might also be interested in helping out

@nikosavola

dan-fritchman, 2025-05-11:
It took quite a while for PyPi to approve it, but we now have: 
https://pypi.org/org/vlsir/

@joamatab  you are invited to it. 
@ThomasPluck - can you share a PyPi username, I'll add you too

ThomasPluck, 2025-05-13:
Sure, I'm [Thomas-Pluck](https://pypi.org/user/Thomas-Pluck/).

dan-fritchman, 2025-05-18:
OK you now have an invitation to https://pypi.org/org/vlsir/.  
And I moved all the vlsir and hdl21 "projects" (PyPi's name for the thing I would call "packages") into there.