Vlsir/Vlsir #105: Time for a Protoc Upgrade?

ThomasPluck, 2025-05-06:
Just getting my PR https://github.com/cascode-labs/Vlsir/pull/1 ready to get #94 merged (as tracked by #93).

I think the protoc version may've slipped during development, decided to leave it an open question if we want to force this version of the protobuf compiler or perhaps take this as an incidental opportunity to upgrade version, or does this like *break* everything??

ThomasPluck, 2025-05-06:
Possibly addressed in #96

dan-fritchman, 2025-05-11:
Don't think an update would break anything, much less everything.  
@growly can weigh in too.  
We generally check in the bindings. So it'd change all their files. But assuming the protobuf compiler change isn't sufficiently different to break those, should be fine.

growly, 2025-05-11:
IIUC the problem is that the tests being introduced rely on a newer protobuf library, is that right?

[bfg](https://github.com/growly/bfg) and other projects that are clients only of the proto schema (as opposed to the bindings) should be fine as long as the schema themselves are not updated with some backwards-incompatible feature. As far as I can tell no such changes are proposed. Great!

(It's worth noting that I had some trouble upgrading `protoc` on my system in tandem with the dependencies `protoc` and `bfg` shared: Abseil, gRPC. It looked like some work to resolve. But this should not force that on anyone.)

ThomasPluck, 2025-05-13:
@growly Yeah, it's more the `protobuf` Python library can't work with newer `protoc` generated bindings. But it sounds like that version bump isn't really issue an issue for anyone, so I'll just roll those changes into my PR.